brgl/libgpiod #121: python toggle_line_value.py example : Toggling GPIOs disables I2C lines.

nickFridays, 2025-01-05:
Hello,  I am using python example toggle_line_value.py  to make sure the bug is not in my code I write for a large app. Please correct me if I am wrong.  I added two lines in toggle_line_value.py.  1. import os  2. print(subprocess.check_output('i2cdetect -y 1', shell=True))
Every time line 2 is executed one or two I2C addresses disappear. 

[image]

brgl, 2025-01-06:
I'm pretty sure this is not a problem with libgpiod but a question that should be asked to the raspberry pi community? You're not saying which pin you're toggling, I would assume it either resets the i2c device or serves as some kind of GPIO mux? I don't know RPi 5 and don't have one.

nickFridays, 2025-01-06:
I also hope it is not a problem with libgpiod because it is practically the only lib working with Pi 5,  thanks to you.
However,  I am using your example python code   toggle_line_value.py  where I believe  you toggle GPIO 5. I tested some other GPIOs. Are you asking about a physical pin # or BCM GPIO # ?
Thank you,

[image]

brgl, 2025-01-06:
I don't know what BCM GPIO is. I don't work with Raspberry Pi. The example code is meant to be generic and show how to use the library. I cannot know which pin you actually want to toggle and what its function is.

nickFridays, 2025-01-06:
Which pin do you toggle in your code ?   
What does 5 mean as a second parameter in your function   toggle_line_value("chip", 5) ?

brgl, 2025-01-06:
It's just an example. It will be pin 5 of chip whatever chip is represented on your system by the `/dev/gpiochip0` character device. You need to adjust the code for your needs. From the output of `gpiodetect` I'd say it will be pin 5 of `gpiochip0 [pinctrl-rp1] (54 lines)` but you will really have more luck talking to the people who know this platform well.

nickFridays, 2025-01-06:
Ok, thanks anyways for you quick response.  I just double checked ,   pininfo  says:   line 5 is GPIO5.

closed 2025-02-20 by brgl