brgl/libgpiod #122: Missing py.typed marker

JPHutchins, 2025-01-16:
The python bindings appear to be typed, but the module is missing the py.typed marker, so it will not be analyzed as such. Mypy provides guidance:

`mypy my_script.py`:
```
my_script.py:1: error: Skipping analyzing "gpiod": module is installed, but missing library stubs or py.typed marker  [import-untyped]
my_script..py:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
Found 1 error in 1 file (checked 1 source file)
```

Cheers,
JP

brgl, 2025-01-17:
Hi! This is already fixed in master and will be released as part of python3-gpiod v2.3.0. I'm just finishing the readthedocs integration as libgpiod was missing proper online docs for a long time. Give it a couple more weeks, please.

JPHutchins, 2025-01-17:
Fixed by https://github.com/brgl/libgpiod/commit/c840e17a9ca05e44ec823b77843ab0e8b88691c3

vfazio, 2025-02-20:
2.3.0 has been published, can we close this?

closed 2025-02-20 by brgl