brgl/libgpiod #125: make install not working from tar.gz files.

YuzuSphynx, 2025-03-07:
Not sure if this is a good place to ask, attempting to install this on a xilinx board and running into issues. Downloading the 2.2.1 tarball, going into the extracted directory, and running various install commands fail. There appear to be two makefiles in the `libgpiod-2.2.1` directory. specifying either with make does not appear to work either due to formatting errors (missing separator).

after cd-ing into the `libgpiod-2.2.1` directory i've tried running:

`make install`
returns: `make: *** No rule to make target `install`. Stop.`

`make -f Makefile.in`
returns: `Makefile.in:15: *** missing separator. Stop.`

`make -f Makefile.am`
returns: `Makefile.am:19: *** missing separator. Stop.`

I haven't tried building it directly, but am almost certainly doing something dumb.

brgl, 2025-03-07:
Hmm maybe I need to improve the docs a bit. In general you want to run:
```
tar -xvf libgpiod-2.2.1.xz
cd ./libgpiod-2.2.1
./configure <whatever flags you need>
make
sudo make install
```

YuzuSphynx, 2025-03-14:
Thanks, not entirely sure but `./configure --enable-tools=yes` complains when i run it... i get an error about some system clock but doing the same on an older version, 1.6.3, yields the expected behavior. after which `sudo make install` runs, i have to also run `sudo ldconfig` before i can call `sudo gpiodetect` (etc) successfully.  

I'm guessing this works on 2.2.1 if i fix the system clock complaint on my end. not sure why 1.6.3 doesn't throw the same error though... I appreciate the help, sorry for the remedial question.

brgl, 2025-03-14:
I cannot help you if you don't at least post the output of `configure`.