brgl/libgpiod #129: What do I need to do to get a Makefile that uses my cross-compile tools?

davenmccalla-diagraph, 2025-04-07:
I got a cross compiler  from here:
`wget -c  'https://mirrors.edge.kernel.org/pub/tools/crosstool/files/bin/x86_64/11.3.0/x86_64-gcc-11.3.0-nolibc-arm-linux-gnueabi.tar.xz'`

Extracted it in my /home directory
`tar -xf x86_64-gcc-11.3.0-nolibc-arm-linux-gnueabi.tar.xz`

I got libgpiod from here:
`wget -c  https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/snapshot/libgpiod-2.1.tar.gz`

Extracted it in my /home directory:
```
tar -xzf libgpiod-2.1.tar.gz
cd libgpiod-2.1

```
Installed a few needed packages:
`apt-get install autoconf autoconf-archive libtool
`

Ran autogen.sh:
```
./autogen.sh --enable-tools=yes --host=arm-linux-gnueabi --prefix=/home
make
make install
```

The shared libraries in /home/lib folder are x86-64 instead of ARM:
```
file /home/lib/libgpiod.so.3.1.0
lib/libgpiod.so.3.1.0: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=2dab4d1992c97d2ab180963e5a6b3e631793e496, with debug_info, not stripped
```
What do I need to do to get a Makefile that uses my cross-compile tools?

davenmccalla-diagraph, 2025-04-07:
Update...
I added the path to the cross-compile tools:
`export PATH=$PATH:/home/gcc-11.3.0-nolibc/arm-linux-gnueabi/bin`

Now it complains that my C compiler doesn't work:
```
checking for arm-linux-gnueabi-gcc... arm-linux-gnueabi-gcc
checking whether the C compiler works... no
configure: error: in `/home/libgpiod-2.1':
configure: error: C compiler cannot create executables
```

I've successfully used the cross-compile tools to build other stuff, so I know it works.

What am I missing now?

brgl, 2025-04-08:
These toolchains are mostly for building the kernel. You may end up missing dependencies as soon as you try to build anything more than the base library. Also: you probably won't have autoconf-archive. I suggest using a toolchain generated by yocto for cross-compiling libgpiod.

davenmccalla-diagraph, 2025-04-08:
Thank you! I should have noticed the "nolibc" in the name of the cross compiler.

I downloaded and extracted a different cross-compiler:
```
wget -c https://releases.linaro.org/components/toolchain/binaries/latest-7/arm-linux-gnueabihf/gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf.tar.xz
tar -xf gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf.tar.xz
```

Added it to the PATH:
`export PATH=$PATH:/home/gcc-linaro-7.5.0-2019.12-x86_64_arm-linux-gnueabihf/bin`

Ran autogen.sh, make, and make install
```
./autogen.sh --enable-tools=yes --host=arm-linux-gnueabihf --prefix=/home
make
make install
```
The shared libraries are now ARM.
```
file ../lib/libgpiod.so.3.1.0
../lib/libgpiod.so.3.1.0: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, BuildID[sha1]=7b46d7fc142c2e3b653fecfd49ec3fdd78263b41, with debug_info, not stripped
```

closed 2025-04-08 by davenmccalla-diagraph