brgl/libgpiod #130: Can you change direction without calling gpiod_line_config_add_line_settings().

davenmccalla-diagraph, 2025-04-14:
After requesting a line:
```
GPIOFiles::GPIOFiles(uint8_t useGpioNum)
{
    gpioNumber = useGpioNum;
    request = NULL;
    settings = NULL;
    line_cfg = NULL;
    chip = NULL;
}

bool GPIOFiles::Initialize()
{
    std::string chip_path  = GPIO_CHIP_PREFIX + to_string(gpioNumber/32);
    unsigned int lines[] = { gpioNumber & 0x1f };
    chip = gpiod_chip_open(chip_path.c_str());
    if (chip) {
        settings = gpiod_line_settings_new();
        if (settings) {
            gpiod_line_settings_set_direction(settings, GPIOD_LINE_DIRECTION_INPUT);
            line_cfg = gpiod_line_config_new();
            if (line_cfg) {
                int ret = gpiod_line_config_add_line_settings(line_cfg, lines, 1, settings);
                if (!ret) {
                    request = gpiod_chip_request_lines(chip, NULL, line_cfg);
                }
            }
        }
    }
    return (request != NULL);
}
```
I want to read the line when it's an input:
```
int8_t GPIOFiles::Read()
{
    enum gpiod_line_value value = gpiod_line_request_get_value(request, gpioNumber & 0x1f);

    return value;
}

```
Change the output value when it's an output:
```
bool GPIOFiles::Write(const char value)
{
    bool success = true;
    enum gpiod_line_value line_value;

    if (value & 1) {
        line_value = GPIOD_LINE_VALUE_ACTIVE;
    } else {
        line_value = GPIOD_LINE_VALUE_INACTIVE;
    }

    if (gpiod_line_request_set_value(request, gpioNumber & 0x1f, line_value) == -1) {
        success = false;
    }
    return success;
}
```
And change the direction as needed:
```
void GPIOFiles::SetDirection(const char direction)
{
    unsigned int lines[] = { gpioNumber & 0x1f };
    enum gpiod_line_direction dir;

    if (direction == OUT) {
        dir = GPIOD_LINE_DIRECTION_OUTPUT;
    } else {
        dir = GPIOD_LINE_DIRECTION_INPUT;
    }

    gpiod_line_settings_set_direction(settings, dir);
    gpiod_line_config_add_line_settings(line_cfg, lines, 1, settings);
    gpiod_line_request_reconfigure_lines(request, line_cfg);

    return;
}
```
**How can I change direction without calling gpiod_line_config_add_line_settings()?** This function allocates memory and the program eventually runs out of memory and dies.
Also, the call to gpiod_line_request_set_value() in Write() always returns a -1, even though it successfully changes the output value.

brgl, 2025-04-17:
There are several issues I see with this code but most importantly: if your program runs out of memory and dies, it's not a problem caused by dynamically allocating memory but by not freeing it. There are official C++ bindings in libgpiod, why are you writing your own, buggy implementation?

davenmccalla-diagraph, 2025-04-17:
The code base that I'm working with already had a C++ class for working with GPIO using sysfs (GPIOFiles), so it made sense to modify the existing functions to use libgpiod instead. My motivation to change from sysfs to libgpiod is because sysfs is deprecated, but when I used libgpiov1 it was much slower than sysfs. This is because you have to release the line and (re)request it in order to change direction. If there's a faster way to change direction with v1, please let me know.

```
bool GPIOFiles::SetDirection(const char direction)
{
    gpiod_line_release(line);
    line = gpiod_chip_get_line(chip, gpioNumber & 0x1f);

    if (line == NULL)
    {
        return false;
    }

    bool success = true;

    if (direction == OUT)
    {
        if (gpiod_line_request_output(line, "GPIOFiles", 0) == READ_FILE_FAILURE)
        {
            //ITW_LOG_ERROR("GPIOFiles SetDirection failed");
            //printf("GPIOFiles SetDirection failed\n");
            success = false;
        }
    }
    else
    {
        if (gpiod_line_request_input(line, "GPIOFiles") == READ_FILE_FAILURE)
        {
            //ITW_LOG_ERROR("GPIOFiles SetDirection failed");
            //printf("GPIOFiles SetDirection failed\n");
            success = false;
        }
    }

    return success;
}
```
The motivation to use libgpiov2 is to be able to change direction without releasing and (re)requesting the line, for increased performance. I didn't know there were C++ bindings, so I used the reconfigure_input_to_output.c example to modify the GPIOFiles class functions. I do have a destructor that frees the memory, but it will only be called when the program ends.
```
GPIOFiles::~GPIOFiles()
{
    gpiod_line_config_free(line_cfg);
    gpiod_line_settings_free(settings);
    gpiod_chip_close(chip);
    gpiod_line_request_release(request);
}
```
It appears that you have to call gpiod_line_request_reconfigure_lines() in order to change direction, and I'm afraid that freeing the memory that it allocates will require me to release and (re)request the line, defeating the purpose for switching to libgpiodv2.

I'll take a look at the C++ binding to see if that will help.

brgl, 2025-04-17:
There's no way to change the direction of a line without releasing it in libgpiod v1 due to kernel uAPI limitations. I'm pretty sure you're doing something wrong here - are you leaking memory? Or are you just inflating one of the structures in some way? IOW: where is the memory allocated in a way that breaks your program?

davenmccalla-diagraph, 2025-04-18:
Yeah, I read several discussions online about libgpiodv1 not being able to change direction due to kernel limitations. But then I learned that those kernel limitations don't exist in my kernel (v5.10). Since the Debian package for libgpiodv2 isn't available for my distro, I had to build it with a cross compiler. You helped me with that a week or two ago. Thank again.

My application is similar to a keyboard matrix, where the rows and columns are "scanned" to see what buttons are being pressed. Only instead of buttons, I have bi-color LEDs that need to flash at >60Hz for them to "appear" to be ON without noticeable flicker. When flashing a LED at a (row, column) location, the other rows and columns need to be inputs. It all worked fine with sysfs, but now I need to migrate to something else.

I was hoping you would recognize most of the code in my buggy C++ implementation, since it mostly came from the **reconfigure_input_to_output.c** example program in your repository. I copied all the steps to request a line and change the direction. The example program only works if you change direction a limited number of times. Otherwise, it would surely run out of memory and die like my program. 

The problem is every time **gpiod_line_config_add_line_settings()** is called, a node is added to the sref_list. In order to prevent sref_list from growing out of control, **gpiod_line_config_free()** has to be called to free line_cfg, only to be immediately reallocated by calling **gpiod_line_config_new()** before calling **gpiod_line_config_add_line_settings()**. This is a little weird and inefficient, but is an acceptable solution.

Here's my updated SetDirection() that doesn't leak memory:

```
void GPIOFiles::SetDirection(const char direction)
{
    unsigned int lines[] = { gpioNumber & 0x1f };
    enum gpiod_line_direction dir;

    if (direction == OUT) {
        dir = GPIOD_LINE_DIRECTION_OUTPUT;
    } else {
        dir = GPIOD_LINE_DIRECTION_INPUT;
    }

    gpiod_line_settings_set_direction(settings, dir);
    gpiod_line_config_free(line_cfg); // free line_cfg...
    line_cfg = gpiod_line_config_new(); // then reallocate line_cfg...
    gpiod_line_config_add_line_settings(line_cfg, lines, 1, settings); // because this function keeps growing line_cfg
    gpiod_line_request_reconfigure_lines(request, line_cfg);

    return;
}
```

brgl, 2025-04-18:
Ok, I understand now. This sounds like something I could improve, I'll take a look next week. We can probably figure out which settings objects are no longer referenced and remove them. In the meantime: can you use `gpiod_line_config_reset()` instead?

brgl, 2025-04-19:
BTW if you don't mind giving me your name and email address, I can give you credit on the patch that will fix the bad behavior.

brgl, 2025-04-24:
@davenmccalla-diagraph  Please try the following series: https://lore.kernel.org/linux-gpio/20250424-line_config_mem_use-v1-0-fa0abdcf0cdf@linaro.org/

closed 2025-04-30 by brgl