brgl/libgpiod #143: tests: tests-line-config.c fails to compile with older glib2.0

vfazio, 2025-08-05:
```
  CC       tests-line-config.o
tests-line-config.c: In function ‘_gpiod_test_func_dont_allow_line_config_to_balloon_out_of_control’:
tests-line-config.c:512:41: error: ‘G_TEST_SUBPROCESS_DEFAULT’ undeclared (first use in this function); did you mean ‘G_TYPE_SUBPROCESS_FLAGS’?
  512 |         g_test_trap_subprocess(NULL, 0, G_TEST_SUBPROCESS_DEFAULT);
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                         G_TYPE_SUBPROCESS_FLAGS
tests-line-config.c:512:41: note: each undeclared identifier is reported only once for each function it appears in

```

```
(venv) vfazio@vfazio4 ~/development/libgpiod $ git blame -L512 tests/tests-line-config.c
c116b6f6 (Bartosz Golaszewski 2025-04-24 14:06:17 +0200 512)    g_test_trap_subprocess(NULL, 0, G_TEST_SUBPROCESS_DEFAULT);
c116b6f6 (Bartosz Golaszewski 2025-04-24 14:06:17 +0200 513)    g_test_trap_assert_passed();
c116b6f6 (Bartosz Golaszewski 2025-04-24 14:06:17 +0200 514) }
```

```
commit c116b6f6f118b983dbd9674612396cc04a883905
Author: Bartosz Golaszewski <bartosz.golaszewski@linaro.org>
Date:   Thu Apr 24 14:06:17 2025 +0200

    tests: add a test-case checking line-config memory usage
    
    Add a test case making sure that repeatedly adding line settings to a
    line-config object doesn't continuously inflate its size until OOM.
    
    Link: https://lore.kernel.org/r/20250424-line_config_mem_use-v1-2-fa0abdcf0cdf@linaro.org
    Signed-off-by: Bartosz Golaszewski <bartosz.golaszewski@linaro.org>
```

Currently, `configure.ac` has:

```
if test "x$with_tests" = xtrue
then
        # For libgpiosim
        PKG_CHECK_MODULES([KMOD], [libkmod >= 18])
        PKG_CHECK_MODULES([MOUNT], [mount >= 2.33.1])

        # For core library tests
        PKG_CHECK_MODULES([GLIB], [glib-2.0 >= 2.50])
        PKG_CHECK_MODULES([GIO], [gio-2.0 >= 2.50])
```

`G_TEST_SUBPROCESS_DEFAULT` wasn't defined until 2.73/2.74:
https://gitlab.gnome.org/GNOME/glib/-/commit/de8672fe0b9f55047fbaee6f425e330cdfc8189f

There are two ways to fix this.

I think the easiest is to swap the enum for `G_TEST_SUBPROCESS_INHERIT_STDIN`:

```diff
vfazio4 /mnt/development/libgpiod/tests :( # git diff .
diff --git a/tests/tests-line-config.c b/tests/tests-line-config.c
index b510e0f..2d26323 100644
--- a/tests/tests-line-config.c
+++ b/tests/tests-line-config.c
@@ -509,6 +509,6 @@ GPIOD_TEST_CASE(dont_allow_line_config_to_balloon_out_of_control)
                return;
        }
 
-       g_test_trap_subprocess(NULL, 0, G_TEST_SUBPROCESS_DEFAULT);
+       g_test_trap_subprocess(NULL, 0, G_TEST_SUBPROCESS_INHERIT_STDIN);
        g_test_trap_assert_passed();
 }
```
```
vfazio4 /mnt/development/libgpiod/tests # ./gpiod-test | grep dont_allow_line_config_to_balloon_out_of_control
ok 57 /gpiod/line-config/dont_allow_line_config_to_balloon_out_of_control
```

This restores support for glib 2.50 so no change to `configure.ac` is required. This allows more host systems to build the package as some distros do not ship libglib2.0-dev 2.74+

The other is to update `configure.ac` to call out a minimal glib2.0 of 2.74:

```diff
(venv) vfazio@vfazio4 ~/development/libgpiod $ git diff configure.ac
diff --git a/configure.ac b/configure.ac
index 9b6c862..8fdb414 100644
--- a/configure.ac
+++ b/configure.ac
@@ -157,7 +157,7 @@ then
        PKG_CHECK_MODULES([MOUNT], [mount >= 2.33.1])
 
        # For core library tests
-       PKG_CHECK_MODULES([GLIB], [glib-2.0 >= 2.50])
+       PKG_CHECK_MODULES([GLIB], [glib-2.0 >= 2.74])
        PKG_CHECK_MODULES([GIO], [gio-2.0 >= 2.50])
 
        AC_CHECK_FUNC([atexit], [], [FUNC_NOT_FOUND_LIB([atexit])])

```

```
configure: error: Package requirements (glib-2.0 >= 2.74) were not met:

Requested 'glib-2.0 >= 2.74' but version of GLib is 2.72.4

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables GLIB_CFLAGS
and GLIB_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

```

vfazio, 2025-08-05:
@brgl let me know which method you prefer and I can submit a patch at some point

brgl, 2025-08-11:
Sorry, I was on vacation, please go with the first approach, I think it's better to not bump the GLib version requirement just for this single flag.

vfazio, 2025-08-16:
No worries. I'm slowly working on this. I have it staged in https://github.com/vfazio/libgpiod/tree/vfazio-fix-test-compile-error but i need to flesh out the commit message

pwithnall, 2025-08-24:
You can use [`#if GLIB_CHECK_VERSION(2,74,0)`](https://docs.gtk.org/glib/func.CHECK_VERSION.html) to conditionally compile a block of code when built against GLib ≥ 2.74.0.

You can also define [`GLIB_VERSION_MIN_REQUIRED`](https://docs.gtk.org/glib/const.VERSION_MIN_REQUIRED.html) and/or `GLIB_VERSION_MAX_ALLOWED` to get compiler warnings if you use GLib APIs outside that range. (libgpiod might already be defining those; I haven’t checked, just thought it would be safer to mention it while I’m here doing this drive-by comment. :)

vfazio, 2025-08-24:
> You can use [`#if GLIB_CHECK_VERSION(2,74,0)`](https://docs.gtk.org/glib/func.CHECK_VERSION.html) to conditionally compile a block of code when built against GLib ≥ 2.74.0.
>
> You can also define [...]

Hey thanks! I'll confirm with Bart how he wants to approach this. Since its only in a test case, the only instance of the issue, and I'm not sure how he feels about ifdef (some maintainers dislike it).

I'll look into GLIB_VERSION_MIN_REQUIRED

brgl, 2025-08-25:
How about requiring a higher GLib version for tests but a lower one for the actual GLib bindings? TBH I don't like the fact that we'd just silently drop parts of the test code.

brgl, 2025-08-26:
Yeah, glib-2.80 is 1.5yo now and seems like major distros already do ship at least this version. Since the gpio-manager was released together with GLib bindings, it will not get into already existing distro releases anyway, only into newer ones. I think it makes sense to keep 2.80 as the minimum version since I like the functionality of the symbols it added and that we're using.

vfazio, 2025-09-09:
fixed in https://github.com/brgl/libgpiod/commit/5e744bf56d84947203b191a79c2e505bcf8d9900

closed 2025-09-09 by vfazio